bluerobotics/ping-viewer #142: Add support for free-running mode

jaxxzer, 2018-06-13:
Ping hardware has a command to enable continuous (non-polled) data stream. Try it out!

patrickelectric, 2018-06-13:
from [here](https://github.com/bluerobotics/ping-protocol-cpp/blob/dev/src/protocol/templates/ping_protocol.json#L120).
```js
            "set_ping_rate_msec": {
                "id": "1004",
                "payload": [
                    {
                        "name": "rate_msec",
                        "type": "u16"
                    }
                ]
            },
```

jaxxzer, 2018-06-13:
No, this one https://github.com/bluerobotics/ping-protocol-cpp/blob/dev/src/protocol/templates/ping_protocol.json#L542

patrickelectric, 2018-06-14:
Our firmware does not have `continuous_start`
I'm receiving `GOT NACK "bad id"`

PR #299 (2018-09-28): Ping: Use continuous start
Use only _all header

Fix #142
closed 2018-09-28 by patrickelectric