bluerobotics/ping-viewer #154: Add ping widget to configure everything possible

patrickelectric, 2018-06-20:
Fix #143

patrickelectric, 2018-06-21:
WIP, using protocol configuration file to generate widgets.
[image]

jaxxzer, 2018-06-27:
This looks like a debug/dev feature rather than something that should be exposed to normal users. Is this the idea?

For #143, I would rather just add a text field to the existing space.

jaxxzer, 2018-06-27:
I mean I think we should at least keep something like the existing config that is compact and concise.

ES-Alexander, 2024-09-26:
This could perhaps be exposed within an "advanced" mode, with a system similar to the one [suggested here](https://github.com/bluerobotics/BlueOS/issues/2771). Agreed that it doesn't belong in the standard user interface.