bluerobotics/ping-viewer #237: Do not erase config data during firmware upgrade

jaxxzer, 2018-08-01:
There is data stored in mcu flash that needs to be retained.

patrickelectric, 2018-08-01:
Are we erasing the eeprom ? Are we saving in the eeprom ?

jaxxzer, 2018-08-01:
There is no eeprom (in any stm32 afaik). It is in flash memory with the program.

patrickelectric, 2018-08-02:
Yes, you are correct. But, in this case we just need to write our variables in the highest memory address correct ?
In my mind, It'll not be possible to update the firmware or to add new variables without breaking the hardcoded address. Like binary compatibility between library versions.

patrickelectric, 2018-08-02:
This is a "must-have" feature ? I don't see that as a high priority thing.

jaxxzer, 2018-08-02:
It is must have, but not immediate priority.

I have dealt with this before. Generally, you dedicate the uppermost region for your non volatile storage, then tell the linker you have that much less flash. Then everything is fine as long as we don't do a mass erase.

patrickelectric, 2018-09-06:
Moved to: https://github.com/bluerobotics/altimeter1-mirror/issues/4

closed 2018-09-06 by patrickelectric