bluerobotics/ping-viewer #241: ci is pushing tags with same name as branch

jaxxzer, 2018-08-02:
Leading to this:
```
error: dst refspec python matches more than one.
error: failed to push some refs to 'https://github.com/bluerobotics/ping-components'
```

patrickelectric, 2018-08-03:
The problem appears not to be from Appveyor.

patrickelectric, 2018-09-06:
Now checking again, it can only be appveyor.
Issue created, waiting feedback: https://help.appveyor.com/discussions/problems/16523-appveyor-is-creating-tags-with-the-same-name-of-the-branches-after-release

jaxxzer, 2018-09-07:
Nice, looks like you got some advice, too.

jaxxzer, 2018-09-11:
I made a mistake, thought I would have to delete a silly named branch, but I can't find this, do I need to clean it out of the github repo?
```
jack@jack-desktop:~/git/pingviewer(addlink)$ git push br addlink:refs/head/addlink
Counting objects: 33, done.
Delta compression using up to 8 threads.
Compressing objects: 100% (32/32), done.
Writing objects: 100% (33/33), 11.08 KiB | 0 bytes/s, done.
Total 33 (delta 8), reused 0 (delta 0)
remote: Resolving deltas: 100% (8/8), completed with 3 local objects.
remote: This repository moved. Please use the new location:
remote:   https://github.com/bluerobotics/ping-viewer.git
To https://github.com/bluerobotics/ping-components
 * [new branch]      addlink -> refs/head/addlink
```

PR #293 (2018-09-11): appveyor: Do not push tags with branch name
Fix #241
commit 939396d6 (2018-09-11): appveyor: Do not push tags with branch name

Fix #241
closed 2018-09-11 by jaxxzer