bluerobotics/ping-viewer #374: Missing ssl functions on windows

jaxxzer, 2018-12-27:
Windows debug release 623d2dd:

```
[12:59:34.971] ping.main[Info]: main.cpp(82) Git version: 623d2dd
[12:59:34.971] ping.main[Info]: main.cpp(83) Git version date: 2018-12-20T20:37:08+00:00
[12:59:34.971] ping.main[Info]: main.cpp(84) Git tag: t0.8.4
[12:59:34.987] ping.main[Info]: main.cpp(85) Git url: https://github.com/bluerobotics/ping-viewer.git
[12:59:34.987] ping.settingsmanager[Debug]: settingsmanager.h(111) "Load darkTheme with:" false
New category registered:  qt.network.ssl
[13:00:04.391] qt.network.ssl[Warning]: ssl\qsslsocket_openssl_symbols.cpp(126) QSslSocket: cannot call unresolved function SSLv23_client_method
[13:00:04.398] qt.network.ssl[Warning]: ssl\qsslsocket_openssl_symbols.cpp(126) QSslSocket: cannot call unresolved function SSL_CTX_new
[13:00:04.404] qt.network.ssl[Warning]: ssl\qsslsocket_openssl_symbols.cpp(126) QSslSocket: cannot call unresolved function SSL_library_init
[13:00:04.404] qt.network.ssl[Warning]: ssl\qsslsocket_openssl_symbols.cpp(126) QSslSocket: cannot call unresolved function ERR_get_error
[13:00:04.404] qt.network.ssl[Warning]: ssl\qsslsocket_openssl_symbols.cpp(126) QSslSocket: cannot call unresolved function ERR_get_error
[13:00:04.404] ping.networktool[Debug]: networktool.cpp(66) Running test version: "t0.8.4"
[13:00:04.419] ping.networktool[Warning]: networktool.cpp(71) GitHub json it's not an array!
[13:00:04.419] ping.networktool[Debug]: networktool.cpp(72) QJsonDocument()
```

patrickelectric, 2019-01-14:
@jaxxzer 	can you confirm that such error does not appears in **release** artifact ?

patrickelectric, 2019-01-14:
It works for both debug and release artifact here in windows.

patrickelectric, 2019-01-14:
@jaxxzer 	Just to update you, my windows installation has the openssl binaries under `C:\Windows\SysWOW64`, after removing it, the error appears.
I'm going to add the .dll files in our release.

PR #376 (2019-01-15): appveyor: Add openssl dlls in deploy folder
Fix #374
commit 5451a96d (2019-01-15): appveyor: Add openssl dlls in deploy folder

Fix #374
closed 2020-11-18 by patrickelectric