bluerobotics/ping-viewer #416: Problems with waterfall scaling

jaxxzer, 2019-01-18:
There is some issue with scaling the waterfall, it looks like profiles are not scaled uniformly. This can be observed in the discontinuities and horizontal bands running across the display beginning around the 2:00 mark in [this](https://github.com/bluerobotics/ping-viewer/files/2774606/20190118-095811025.txt) log.

[image]

patrickelectric, 2019-01-19:
After playing the log several times I was unable to find this issue,
you can see my screenshots here during all the execution:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Can you take a screenshot in the exact moment ?

jaxxzer, 2019-01-19:
[Here](https://gfycat.com/IlliteratePlushAmericansaddlebred), I seek to 2:30 and push 'R' to refresh the display. You can see as the scan range becomes longer, the past data is not scaled uniformly. It is like sections of it are moving like tectonic plates, and sections of data slide up and under other data. Some of the data is lost and this appears as straight horizontal lines across the display where the data breaks.

Also, the very top portion of the data does not really shrink at all.

jaxxzer, 2019-01-19:
Using [`ping1D-simulation.py`](https://github.com/jaxxzer/ping-python-1/blob/pingsim/brping/ping1d-simulation.py) I cannot reproduce this sort of behavior, everything appears to scale appropriately, I'm a bit surprised about that result.

[image]

[image]