bluerobotics/ping-viewer #444: Cannot edit scan start/stop

jaxxzer, 2019-01-29:
There is some problem editing scan start/stop on v1.0.0, but the speed of sound works correctly.

jaxxzer, 2019-01-29:
I've spent about an hour on this and have not fixed it.

|- -|- v3.24 -|- v3.26 -|
|---|---|---|
| t0.9.1 | working | working |
| v1.0.0 | not working | not working |

It appears related to the [focus](https://github.com/bluerobotics/ping-viewer/blob/master/qml/PingTextField.qml#L32-L44) of PingTextField. It's interesting that [speed of sound](https://github.com/bluerobotics/ping-viewer/blob/master/qml/MainMenu.qml#L67) works and[ scan range](https://github.com/bluerobotics/ping-viewer/blob/master/qml/MainMenu.qml#L177) does not.

Also #436 works without issue.

patrickelectric, 2019-01-29:
[image]
You can see in the gif of https://github.com/bluerobotics/ping-viewer/pull/437 that it was working before the rebase and commit correction.
Will try to see what is missing from the original work.

PR #449 (2019-01-29): Correct PingTextField behavior 
Fix #444 and #447
closed 2019-01-29 by patrickelectric