bluerobotics/ping-viewer #492: Ping nack: `bad continuous request id: 6`

jaxxzer, 2019-04-05:
Seen on master + 3.26. May be a firmware bug...

jaxxzer, 2019-04-05:
[image]

patrickelectric, 2019-04-05:
general request requesting general request ?
lol

patrickelectric, 2019-04-08:
We are not requesting id 6, or NACK message does not specify why the request failed. 
This is a list of requested messages before the NACK message appears:
```
>>> Requesting: 1215
>>> Requesting: 1205
>>> Requesting: 1300
>>> Requesting: 1200
>>> Requesting: 1201
>>> Requesting: 1203
>>> Requesting: 1215
>>> Requesting: 1205
>>> Requesting: 1206
>>> Requesting: 1214
>>> Requesting: 1213
>>> Requesting: 1202
>>> Requesting: 1205
>>> Requesting: 1214
>>> Requesting: 1213
>>> Requesting: 1202
>>> Requesting: 1205
```
I believe that is a firmware issue, what should we do related to it ?

patrickelectric, 2019-04-08:
Yes, the sensor replies NACK for [ID 1300](https://github.com/bluerobotics/ping-protocol#1300-profile).

jaxxzer, 2019-04-11:
Moved to ping-firmware repository: https://github.com/bluerobotics/ping-firmware/issues/2

closed 2019-04-11 by jaxxzer