bluerobotics/ping-viewer #518: ping360 future features

jaxxzer, 2019-06-13:
Dropping brainstorm ideas here:

- NMEA input to display gps coordinates on sonar display mouseover
- Angular resolution adjustment (increase scan speed)
- Sector scan mode
- side scan mode
- firmware update
- Horizontal/vertical flip display (for upside down mounting)
- Rotating display (mounting offset)
- Rotating display (compass input)
- Zoom controls

kklemens, 2019-06-13:
- Receiver Gain Control (slider would be preferred)

- Origin Offset

patrickelectric, 2019-07-03:
@kklemens 	We only have 3 states for gain control, so this right now is a combobox with Low, Normal, High gain;

BoKoppel, 2019-12-07:
Distant Ping360 feature for sidescan use:
-Lock scanning 90 degree sideway
-Create sidescan files from positioning and heading, ie standard XTF format