bluerobotics/ping-viewer #551: Avoid bad operations with ping360 bootloader

jaxxzer, 2019-07-17:


patrickelectric, 2019-07-17:
can you described what is a bad operation ? This issue is not clear.

jaxxzer, 2019-07-17:
We should not send any serial communications to the bootloader that is not in compliance with the bootloader protocol.

patrickelectric, 2019-07-17:
how can we ensure that ? We are not doing it right now, are we ? Besides that, it's not possible for ping-viewer to identify if the device is in bootloader after it's connected, this is more for the user that ping-viewer itself, if the sensor is in bootloader its because we put in that state.

jaxxzer, 2019-07-17:
> We are not doing it right now, are we ?

no

>  it's not possible for ping-viewer to identify if the device is in bootloader after it's connected

It is possible to identify the bootloader by communicating with it.

> how can we ensure that ?

We can communicate with the bootloader, and if we get a response, wait for it to exit before sending ping-messages.

> if the sensor is in bootloader its because we put in this state or the user did that somehow.

This is not true. The bootloader is active automatically on boot. The user can connect the sonar while the application is open, and we send ping-message to the bootloader. This is the issue.

patrickelectric, 2019-07-17:
If device manager, are we going to start detect for sensors using
ping-protocol and all available bootloaders that we support ?
This issue is not clear for me, and less clear how we are going to fix that
if it's an issue.

>This is not true. The bootloader is active automatically on boot. The user
can connect the sonar while the application is open,
That's what I'm pointing, the user should not do that. I can't think how we
are going to detect the bootloader state, and if all our bootloaders
support his kind of feature.

jaxxzer, 2019-07-24:
The need to not communicate is a shortcoming of the bootloader. We will correct the bootloader behavior.

closed 2019-07-24 by jaxxzer