bluerobotics/ping-viewer #576: Waveform is plotting data beyond extents

jaxxzer, 2019-07-22:
[image]

patrickelectric, 2019-07-26:
I'm not able to replicate it, can you provide a step-by-step ?

jaxxzer, 2019-08-05:
1. download continuous release from master build
1. run it
1. reset application settings
1. connect to ping360 device

It looks like this:
[image]

jaxxzer, 2019-08-05:
After manually deleting the settings file, this is what I get: 
[image]

jaxxzer, 2019-08-05:
You can apply this diff to demonstrate the problem:
```
diff --git a/src/link/ping360simulationlink.cpp b/src/link/ping360simulationlink.cpp
index f218cca..2700b37 100644
--- a/src/link/ping360simulationlink.cpp
+++ b/src/link/ping360simulationlink.cpp
@@ -16,8 +16,8 @@ void Ping360SimulationLink::randomUpdate()
     static const float numberOfSamples = 1200;
     static const int angularResolution = 400;
 
-    const float stop1 = numberOfSamples / 2.0 - 10 * qSin(counter / 10.0);
-    const float stop2 = 3 * numberOfSamples / 5.0 + 6 * qCos(counter / 5.5);
+    //const float stop1 = numberOfSamples / 2.0 - 10 * qSin(counter / 10.0);
+    //const float stop2 = 3 * numberOfSamples / 5.0 + 6 * qCos(counter / 5.5);
 
     static ping360_device_data deviceData(numberOfSamples);
     deviceData.set_mode(0);
@@ -30,12 +30,10 @@ void Ping360SimulationLink::randomUpdate()
 
     for (int i = 0; i < numberOfSamples; i++) {
         float point;
-        if (i < stop1) {
-            point = 0.1 * (qrand()%256);
-        } else if (i < stop2) {
-            point = 255 * ((-4.0 / qPow((stop2-stop1), 2.0)) * qPow((i - stop1 - ((stop2-stop1) / 2.0)), 2.0)  + 1.0);
+        if(i >255 && i < 400) {
+            point = 255;
         } else {
-            point = 0.45 * (qrand()%256);
+            point = i%256;
         }
         deviceData.set_data_at(i, point);
     }
```

[image]

closed 2019-08-05 by jaxxzer