bluerobotics/ping-viewer #577: Weird delay after leaving device manager

jaxxzer, 2019-07-22:
After selecting a device from device manager there is a 3~5 second pause after the window exits before any sensor activity or visualizer activity happens. This even affects the simulation links.

patrickelectric, 2019-07-30:
I'm unable to replicate it in master branch, can you send a gif or a step-step to replicate it ?

Williangalvani, 2019-07-31:
I see a much shorter pause here:
[image]

I can see it via USB or Ethernet. It seems to last ~1s here.

patrickelectric, 2019-07-31:
Well, here is less than 1 to start to receive data from the sensor and to configure everything necessary before the sensor starts talking. I believe that such time is acceptable. I was not able to replicate the 5~10s delay.

patrickelectric, 2019-07-31:
This is probably a issue related to the wait of the detection thread:
https://github.com/bluerobotics/ping-viewer/blob/f893d00f5cbd1380bf9f84a244bf3c3d9362aa28/src/devicemanager/devicemanager.cpp#L67

closed 2019-07-31 by patrickelectric

jaxxzer, 2019-08-05:
This solved it for me :+1: