bluerobotics/ping-viewer #600: ping360: make range scale more dynamic and user-friendly

jaxxzer, 2019-07-25:
It will be nice if the range scale on ping360 behaved the same as the depth scale on ping1d.

The ping1d adjusts **both** the number of increments _and_ the increment value to make the scale easy for humans to interpolate between tick marks.: https://github.com/bluerobotics/ping-viewer/blob/master/qml/DepthAxis.qml#L16

For example, when the range is 30, the labels are 7.5, 15, 22.5, 30:
[image]

I think it would be more natural and easy to interpret the data if the labels were [5, 10, 15, 20, 25, 30], or [10, 20, 30].

jaxxzer, 2019-07-25:
https://giphy.com/gifs/S9o3EXFHwpG0xrzOsU

patrickelectric, 2019-10-11:
Duplicate of #586

closed 2019-10-11 by patrickelectric