bluerobotics/ping-viewer #624: Menu Display on 4k

brianhBR, 2019-07-30:
## Summary
Version: commit: 6ed95bc

Operating System: Linux (Ubuntu)

Control menu is not displayed properly, probably due to the 4k display. 

How to reproduce it: Open PingViewer > Click in the settings icon

## Additional information
[image]

patrickelectric, 2019-07-30:
Thanks Kevin, indeed this is a problem that we got today after some rework in the submenu item. You can check the status here: https://github.com/bluerobotics/ping-viewer/issues/623
We also have a PR to fix that issue, I'll ping you after the next release with this fix.
https://github.com/bluerobotics/ping-viewer/pull/626

closed 2019-07-30 by patrickelectric

patrickelectric, 2019-07-30:
@brianhBR 	you can check our last release, it'll not have the problem descried in this issue: https://github.com/bluerobotics/ping-viewer/releases/tag/continuous