bluerobotics/ping-viewer #625: Device detection under windows takes more than a few seconds to find device

patrickelectric, 2019-07-30:


patrickelectric, 2019-07-31:
@brianhBR 	It took 2~3 seconds here in my local machine with windows and the last version of our continuous build.
https://s3.amazonaws.com/downloads.bluerobotics.com/PingViewer/builds/continuous/pingviewer_release.zip

Tested with both Ping1D and Ping360.

rjehangir, 2019-07-31:
@brianhBR and @patrickelectric - How many COM ports do you have listed in Windows?

patrickelectric, 2019-07-31:
[image]

brianhBR, 2019-07-31:
There are only a few that are active but they are up pretty high in number. Does the program go through all of the port numbers or just the ones that are active?

[image]

patrickelectric, 2019-07-31:
@brianhBR 	It'll scan all available ports that are described as serial by the OS.
Can you share the interface LOG while scanning your devices ?

brianhBR, 2019-07-31:
[20190731-113925741.txt](https://github.com/bluerobotics/ping-viewer/files/3453440/20190731-113925741.txt)

patrickelectric, 2019-07-31:
@brianhBR 	thanks, indeed this is a really weird behavior that you are facing.
```
[11:39:32:907] ping.protocol.protocoldetector[Debug]: UDP socket disconnected.
[11:39:32:973] ping.protocol.protocoldetector[Debug]: Waiting port to open..  0 "COM18"
[11:39:32:973] ping.protocol.protocoldetector[Debug]: Probing Serial "COM18" 115200
[11:39:32:973] ping.protocol.protocoldetector[Debug]: Port is open
[11:40:03:056] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:107] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:172] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:223] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:274] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:338] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:389] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:440] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:491] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:542] ping.protocol.protocoldetector[Debug]: ""
[11:40:03:596] ping.protocol.protocoldetector[Debug]: Waiting port to open..  0 "COM18"
[11:40:03:596] ping.protocol.protocoldetector[Debug]: Probing Serial "COM18" 9600
[11:40:03:596] ping.protocol.protocoldetector[Debug]: Port is open
[11:40:33:673] ping.protocol.protocoldetector[Debug]: ""
[11:40:33:738] ping.protocol.protocoldetector[Debug]: ""
[11:40:33:789] ping.protocol.protocoldetector[Debug]: ""
[11:40:33:840] ping.protocol.protocoldetector[Debug]: ""
[11:40:33:892] ping.protocol.protocoldetector[Debug]: ""
[11:40:33:942] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:005] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:056] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:107] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:157] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:222] ping.protocol.protocoldetector[Debug]: Waiting port to open..  0 "COM1"
[11:40:34:273] ping.protocol.protocoldetector[Debug]: Waiting port to open..  1 "COM1"
[11:40:34:273] ping.protocol.protocoldetector[Debug]: Probing Serial "COM1" 115200
[11:40:34:273] ping.protocol.protocoldetector[Debug]: Port is open
[11:40:34:356] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:406] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:457] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:508] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:559] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:610] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:661] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:712] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:763] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:813] ping.protocol.protocoldetector[Debug]: ""
[11:40:34:943] ping.protocol.protocoldetector[Debug]: Waiting port to open..  0 "COM1"
[11:40:34:994] ping.protocol.protocoldetector[Debug]: Waiting port to open..  1 "COM1"
[11:40:34:994] ping.protocol.protocoldetector[Debug]: Probing Serial "COM1" 9600
[11:40:34:996] ping.protocol.protocoldetector[Debug]: Port is open
[11:40:35:073] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:124] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:174] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:225] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:276] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:327] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:378] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:429] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:480] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:530] ping.protocol.protocoldetector[Debug]: ""
[11:40:35:663] ping.protocol.protocoldetector[Debug]: Waiting port to open..  0 "COM19"
[11:40:35:724] ping.protocol.protocoldetector[Debug]: Waiting port to open..  1 "COM19"
[11:40:35:789] ping.protocol.protocoldetector[Debug]: Waiting port to open..  2 "COM19"
[11:40:35:841] ping.protocol.protocoldetector[Debug]: Waiting port to open..  3 "COM19"
[11:40:35:905] ping.protocol.protocoldetector[Debug]: Waiting port to open..  4 "COM19"
[11:40:35:956] ping.protocol.protocoldetector[Debug]: Waiting port to open..  5 "COM19"
[11:40:36:022] ping.protocol.protocoldetector[Debug]: Waiting port to open..  6 "COM19"
[11:40:36:088] ping.protocol.protocoldetector[Debug]: Waiting port to open..  7 "COM19"
[11:40:36:140] ping.protocol.protocoldetector[Debug]: Waiting port to open..  8 "COM19"
[11:40:36:191] ping.protocol.protocoldetector[Debug]: Waiting port to open..  9 "COM19"
[11:40:36:191] ping.protocol.protocoldetector[Debug]: Couldn't open port "COM19"
[11:40:36:196] ping.protocol.protocoldetector[Warning]: Fail to open serial port: QSerialPort::PermissionError
[11:40:36:255] ping.protocol.protocoldetector[Debug]: Waiting port to open..  0 "COM19"
[11:40:36:255] ping.protocol.protocoldetector[Debug]: Couldn't open port "COM19"
[11:40:36:322] ping.protocol.protocoldetector[Debug]: Waiting port to open..  0 "COM27"
[11:40:36:374] ping.protocol.protocoldetector[Debug]: Waiting port to open..  1 "COM27"
[11:40:36:374] ping.protocol.protocoldetector[Debug]: Probing Serial "COM27" 115200
[11:40:36:380] ping.protocol.protocoldetector[Debug]: Port is open
[11:40:36:409] ping.protocol.protocoldetector[Debug]: "BR\x06\x00\x04\x00\x02\x00\x02""C\x03\x00\x04\x00\xEC\x00"
[11:40:36:409] ping.protocol.protocoldetector[Debug]: Detect new device: 
```

It appears that the bluetooth adapters that you are using (from your screenshot), are making ping-viewer to stuck waiting for something from the OS. I'll try to get some bluetooth adapters here to test this behavior.

brianhBR, 2019-07-31:
You're right! I unplugged the adapter and it's connecting much faster. Thanks!

patrickelectric, 2019-08-14:
@brianhBR 	let me know if this fix your problem:
https://github.com/bluerobotics/ping-viewer/releases/tag/continuous-fix_brian_bt_problem

closed 2019-08-15 by patrickelectric