bluerobotics/ping-viewer #650: Ping360 visualizer works at 7Hz

patrickelectric, 2019-08-06:
This bug appears to affect only windows.
In general, it appears that everything runs slower on windows, more investigation is necessary.

jaxxzer, 2019-08-06:
There are at least two problems here:
1. The 2.5M and some other baudrates are not working on Windows. When 2.5M is selected, the communication actually happens at 115200, even when the ping-viewer indications are that it is going at 2.5M. (2M does work correctly on windows).
2. The viewer-side latency and jitter are both huge on windows (need to do more tests to find root cause of latency):
[image]

jaxxzer, 2019-08-06:
1. is a quick and easy fix @patrickelectric, we should remove the 2.5M baudrate option (the product document will specify 2M maximum).

patrickelectric, 2019-08-06:
2.5M is the problem ? It works in higher frequencies with 2M ?

jaxxzer, 2019-08-06:
@patrick, please remove the 2.5M option:
1. It is not allowed according to the product technical specification.
1. It does not work on windows.

jaxxzer, 2019-08-06:
> 2\. The viewer-side latency and jitter are both huge on windows (need to do more tests to find root cause of latency):

I have some fixes lined up for 2.

PR #658 (2019-08-07): ping360: max valid baudrate is 2Mbaud
fix #650
closed 2019-08-07 by patrickelectric