bluerobotics/ping-viewer #659: Is Ping 360 available for download?

wangshujun-tj, 2019-08-08:
## Summary
_A clear and concise description of what you want to happen._

## Additional information
_Add any other context or screenshots/mockups about the feature request here._

patrickelectric, 2019-08-08:
Hi @wangshujun-tj, there is no public information about this yet. 
What are you aiming for ?

wangshujun-tj, 2019-08-09:
HI， Patrick José Pereira：

        I'm making an air version compatible with ping1d, reserving an interface for the Servo, and if it's convenient, I can make an air version compatible with ping360.

wangshujun@tom.com
 
From: Patrick José Pereira
Date: 2019-08-08 20:02
To: bluerobotics/ping-viewer
CC: wangshujun-tj; Mention
Subject: Re: [bluerobotics/ping-viewer] Is Ping 360 available for download? (#659)
Hi @wangshujun-tj, there is no public information about this yet.
What are you aiming for ?

patrickelectric, 2019-08-09:
Hi @wangshujun-tj, ping-viewer will be compatible with Ping1D, so if your sensor works with Ping protocol under Ping1D messages, ping-viewer will detect and run the necessary viewer for Ping1D. I can't say much about out Ping360 development right now, but you should not worry about it if your sensor will be something like Ping1D but for air.
If you want to check if the master branch will be compatible with your sensor, you just need to compile or download our last continuous/test versions of PingViewer.

jaxxzer, 2019-08-09:
@wangshujun-tj You can see the protocol here: 

http://docs.bluerobotics.com/ping-protocol/pingmessage-ping360/

closed 2019-08-09 by jaxxzer