bluerobotics/ping-viewer #667: UI adjustments

jaxxzer, 2019-08-09:
The engineers at the lab have some requests on adjustments to the UI

@brianhBR please list them here.

brianhBR, 2019-08-20:
UI Comments:
=============
We recently went through the UI with the goal to see if there were any things that we thought might be better laid out. We are totally open to debating this and want to hear your thoughts. If anything is unclear I'd be happy to post a terribly drawn sketch. 

Also, I think it would be cool to add more tool tips to menu icons and sonar settings. I think the overall goal is that this software be as close to not needing a manual as possible.

Controls Page (Top menu with Arrow):
-----------------------------------
- Replace the wrench with an arrow similar to the one used to open the menu. It should point down with the advanced controls menu closed and up with it open
- Move the firmware update icon to it's own sub-menu on the gear settings menu. Also, maybe we replace the icon with something more like an update icon? [image]

- Add a dropdown to set mount orrientation (either head up or head down). This is to replace flip vertically or flip horizontally on the display page. 
- Move Auto Baud and baud setting to Device manager
- Add "reset to defaults" button
- Drop Angle offset - I don't think anyone will need this and it is a little confusing
- Try centering sliders and increase the padding on the menu? It looks a little off balance. When Receiver gain is set to medium it almost touches the border.

Settings Page (Gear): 
---------------------
- Change order to Device Manager, Display settings, Firmware update
- Add an option to hide the A-Scan on the display settings. Its very cool to have the A-Scan but in situations with limited screen area the circular plot is the priority.

Device manager:
------------------
General:
- 3 visually similar pages: Auto, Manual, and Demo. They should each have the same layout with the BR logo at the top and some feedback about the connection status under the logo
- I'd propose that we drop the scrolling banner (talking with dolphins, etc.) and use moving elipses (. , .., ...) after waiting for devices. The scrolling banner is fun but might be frustrating if the user is having connection issues.
- Add the device manager logo to the top left corner of the window (helps to teach user that that logo gets you to the device manager)

Auto page:
- We should move the simulations to the demo page so they don't distract from actual connected devices
- The green and red lights are a cool but might be better. I've had an active connection while it shows red and no connection with it showing green. I think they should be rock solid reliable or maybe we drop it?

Manual page:
- Visually similar to the auto page
- After clicking connect, it should say connecting ..., then connected and/or  we list previously connected devices with connection status (red/green) in a similar way as the auto page

Demo page:
- This is the page for potential customers to test out the interface. It would be cool to have actual sonar data recordings to playback here.

patrickelectric, 2019-09-20:
I believe that mostly of the points are covered in PRs, about the icon:
The idea of the main menus is to be symmetric and able to spin for rotation animations, as we have for the others icons already, I did some work and created a different icon for the firmware update here:
[image]

patrickelectric, 2019-09-23:
The "Demo" idea was moved to manual connection.

closed 2019-09-23 by patrickelectric