bluerobotics/ping-viewer #692: M5: Detection takes a long time to get Ping360 under sensor as dhcp server

patrickelectric, 2019-08-21:
> "Initial "Waiting for devices..." screen - when connected through a router finds an
IP address reasonably quickly but connected on a direct cable I measured the
wait time to be 110 seconds before recognising a device on a 169.254.xxx.xxx
address. If during the wait time either the manual or cancel buttons are clicked
then you get the Windows "not responding" message in the title bar and if you
try to close the program it crashes out.
With the "Discovery" message implemented and sent to a broadcast address of
255.255.255.255 then you should get an instant response from the sonar no
matter what IP address it is on. Is the problem related to your port scanning
routine attempting to write to a non broadcast address first and then suffering
one or more network timeout events before moving on?"

patrickelectric, 2019-08-23:
@Williangalvani can you update the status of this issue ?

Williangalvani, 2019-08-23:
I cannot replicate it under Windows, it worked flawlessly every way I tried it.

Under Linux I cannot get an IP from the DHCP server at all, which doesn't depend on ping-viewer at all. @jaxxzer was going to do some more research on this.

patrickelectric, 2019-08-23:
I'm moving the PR to invalid since we are unable to replicate it. Waiting for future updates.

closed 2019-11-15 by patrickelectric