bluerobotics/ping-viewer #741: Antialiasing Slows Scan Rate on Mac

brianhBR, 2019-09-05:
---
## Summary
Version: 2.0.2

Operating System:Mac (Mojave)

What is happening: The scan rate for ping360 is dramatically reduced from ~30 to ~20 FPS when antialiasing is on.

How to reproduce it: With a Ping360 attached open PingViewer > open display settings and toggle antialiasing.

jaxxzer, 2019-09-05:
We should disable the setting by default or remove it altogether.

I think smooth data should probably be disabled by default also.

jaxxzer, 2019-09-05:
I was not able to reproduce it on linux, scan rate was not impacted, and cpu load did not change significantly.

patrickelectric, 2019-09-19:
@brianhBR just to make sure, I can see an improvement of 6~8Hz on mac when dissabling the anti-aliasing, is that what are you talking about ?

PR #755 (2019-09-20): Remove default antialiasing when running under mac
Fix #741
closed 2019-09-20 by patrickelectric