bluerobotics/ping-viewer #747: ping360: reduce timeout for profile request

jaxxzer, 2019-09-13:
If we have not received a `ping360_data` message in the last 30 seconds, the timeout should be set to 4 seconds.

When we receive a `ping360_data` message, we can assume the motor is already in position and reduce the timeout to 100ms or so.

We have a problem today, where the ping360 will stop scanning for ~10 seconds when there is a hiccup in communication:
[image]

patrickelectric, 2019-09-13:
Not respecting the WCS of 4s that is in the datasheet can result in problems in communication channels that are not full duplex, If I'm not mistaken we have already discussed that and that's why we have a 4.2s timeout, 200ms for the network delay and 4s for sensor WCS.

Williangalvani, 2019-09-16:
I understand that @jaxxzer 's solution is taking that into account.
The sensor only re-homes if it was idle for 30s. If we just got data from it, we know it is not going to re-home an can use a smaller timeout instead.

We need to be careful and take into consideration the range, baudrate, and possibly the current head position.

But the current behavior of stopping for ~10 seconds because of a dropped packet is not acceptable.

PR #753 (2019-09-19): ping360: Improve timeout logic
Use the angular speed plus the network delay to calculate the next probable timeout
Go back to the 4.2s WCS timeout after a 30s communication loss

Fix #747
closed 2019-09-19 by patrickelectric