bluerobotics/ping-viewer #803: Crash when shrinking window too small

NickNothom, 2019-10-08:
## Summary
Version: 2.0.4

Operating System: Windows 10

What is happening: PingViewer crashes when the window is shrunk to the size of the depth reading in height or width. 

How to reproduce it: Open PingViewer with Ping1D, shrink the window in width or height until it is the size of the depth reading text label. 

## Additional information
https://streamable.com/3v34e

Log:
https://drive.google.com/open?id=1W0MiRy91xgu4U6akravUPvkFGUfYpX6M

patrickelectric, 2019-10-08:
Hi @NickNothom can you test [this version](https://github.com/bluerobotics/ping-viewer/releases/download/continuous-fix_small_window/pingviewer_release.zip) ? It should fix your issue.

NickNothom, 2019-10-08:
That version does not fix the issue for me

jaxxzer, 2019-10-08:
:/

NickNothom, 2019-10-08:
Let me know if I can provide any more helpful information

jaxxzer, 2019-10-08:
Maybe this has something to do with the waveform plot on the right disappearing rather than the depth text (notice these two things happen coincidentally in @NickNothom's screenshot). @NickNothom can you make the depth text as small as possible by scroll wheeling over it, then try the same test.

patrickelectric, 2019-10-08:
@NickNothom Please run the debug build in the continuous session and report the crash report.

patrickelectric, 2019-10-09:
You can get the [debug build here](https://github.com/bluerobotics/ping-viewer/releases/download/continuous-fix_small_window/pingviewer_debug.zip)

patrickelectric, 2019-10-10:
@NickNothom I just tested our last continuous version and its working fine unders windows:
[image]

I was able to minimize to the 0x0 size.

jaxxzer, 2019-10-11:
Works ok on linux.

NickNothom, 2019-10-11:
0806d78 and 63f9308 are both still crashing for me on Windows 10. 

I can't run the debug build right now because:

`qrc:/main.qml:-1 File was compiled ahead of time with an incompatible version of Qt and the original file cannot be found. Please recompile`

It seems related to the depth text label to me, because if I move that label around it changes where the crash occurs. You'll see in the second example here that if I quickly make the windows smaller than the text label it's okay, but when I slowly resize it to be the same size as the text label, it crashes. 

https://streamable.com/y6hj3

patrickelectric, 2019-10-14:
Hi Nick, yes the windows debug build is broken: https://github.com/bluerobotics/ping-viewer/issues/812
I'll try to test one more time here with the exactly steps that you did in your video next time that I boot under windows.

patrickelectric, 2019-11-05:
Hi @NickNothom, I was able to replicate your issue in a new Windows installation and the last continuous build of appveyor.

patrickelectric, 2019-11-05:
@NickNothom Can you test this build ?
https://ci.appveyor.com/api/buildjobs/i6wnkfwm4900cvfh/artifacts/pingviewer_release.zip

NickNothom, 2019-11-05:
Fixed for me! 🎉

PR #804 (2019-10-08): Improve and fix ValueReadout behavior 
Fix #803
PR #817 (2019-11-06): ValueReadout: Check for each variable to avoid multiple calls
- Add test for margin and maxSize values

Fix #803
closed 2019-11-06 by jaxxzer