bluerobotics/ping-viewer #812: Windows debug build if failing to run

patrickelectric, 2019-10-10:
It appears a mismatch between kcrash and ping-viewer Qt build versions

samtempledev, 2020-05-18:
Has anyone had any luck with the Debug build in Windows? 'KCrash' file not found line 10 main.cpp

patrickelectric, 2020-05-18:
Hi @samtempledev we are not deploying anymore kcrash, this is something that we want to fix, but not in a near future.

Are you having any problems with ping-viewer ?

samtempledev, 2020-05-18:
Hi @patrickelectric yes the debug project will not build in Qt because of these lines:

#if defined(QT_DEBUG) && defined(Q_OS_WIN)
#include <KCrash>
#endif

Cannot open include file: 'KCrash':No such file or directory

samtempledev, 2020-05-18:
@patrickelectric also this issue:

D:\GitRepos\ping-viewer\src\waterfall\gradientscale.h:37: warning: Property declaration waterfallGradient has no READ accessor function or associated MEMBER variable. The property will be invalid.

patrickelectric, 2020-05-18:
Ok, you are building the source code, now it's much clear. 
KCrash should be compiled and provided as primary buildstep, this is done via [craft compile system](https://community.kde.org/Craft). If you are doing your own test build for development debug, you can remove the kcrash includes, they are only necessary in distributed debug binaries.

samtempledev, 2020-05-18:
@patrickelectric I have successfully ran a build of the Release configuration but upon opening pingviewer.exe i get DLL issues:
Qt5Charts.dll
Qt5Widgets.dll
Qt5QuickControls2.dll
Qt5Quick.dll

All seem to be not found

patrickelectric, 2020-05-18:
Please follow our [build instructions here](https://docs.bluerobotics.com/ping-viewer/build-instructions/), for windows development, I highly recommend to use QtCreator, since it'll run the binary with the correct environment variables for you Qt Installation.

samtempledev, 2020-05-18:
@patrickelectric I have followed those instructions line by line and still no luck with building a debug or release configuration, this is in Windows 10 using Qt Creator and Qt 5.12.0

patrickelectric, 2020-05-18:
What problem are you facing with release build ?
Are you running the binary from QtCreator IDE or via terminal ? Try to do everything in QtCreator if you can.

samtempledev, 2020-05-18:
It seems that when I run it from the Green Play button it works fine, but trying to run it from the .exe file in the folder system it gets missing dll issues for these files:

Qt5Charts.dll
Qt5Widgets.dll
Qt5QuickControls2.dll
Qt5Quick.dll

patrickelectric, 2020-05-18:
Because you need to deploy your applicant, I would recommend to get some some book or read the Qt documentation about C++ development using Qt.
For the deployment documentation: https://wiki.qt.io/Deploy_an_Application_on_Windows