bluerobotics/ping-viewer #816: Can't open the release pingviewer.exe in win10

newJames, 2019-10-22:
I try to open the pingviewer.exe to receive the sonar data,but when I cliked the .exe, the software icon show in the win toolbar and the starting bluerobotice icon showed on the center of the screen ,but there were not any new window show in the os, i can open it on the other win10.

patrickelectric, 2019-10-22:
Hi @newJames. 
Do you have all video drivers updated ?
Also, can you share the last GUI log available ? you can check where it's [here](https://docs.bluerobotics.com/ping-viewer/logging-and-user-files/#gui-logs).

newJames, 2019-10-23:
> Hi @newJames.
> Do you have all video drivers updated ?
> Also, can you share the last GUI log available ? you can check where it's [here](https://docs.bluerobotics.com/ping-viewer/logging-and-user-files/#gui-logs).

Thanks for your answer,the GUI log print the fowlling msg:

[10:14:47:273] ping.settingsmanager[Debug]: "Load darkTheme with:" false
[10:14:47:277] ping.settingsmanager[Debug]: "Load debugMode with:" false
[10:14:47:412] ping.settingsmanager[Debug]: "Load replayMenu with:" false
[10:14:47:414] ping.settingsmanager[Debug]: "Load enabledCategories with:" 0
[10:14:47:414] ping.settingsmanager[Debug]: "Load logScrollLock with:" true
[10:14:47:414] ping.settingsmanager[Debug]: "Load logScrollLock with:" true
[10:14:47:426] ping.settingsmanager[Debug]: "Load debugMode with:" false
[10:14:47:426] ping.settingsmanager[Debug]: "Load debugMode with:" false
[10:14:47:427] ping.settingsmanager[Debug]: "Load debugMode with:" false
[10:14:47:430] ping.settingsmanager[Debug]: "Load debugMode with:" false
[10:14:47:430] ping.settingsmanager[Debug]: "Load replayMenu with:" false
[10:14:47:430] ping.settingsmanager[Debug]: "Load darkTheme with:" false
[10:14:49:329] ping.settingsmanager[Debug]: "Load enabledCategories with:" 0
[10:14:49:440] default[Warning]: FirmwareUpdate.qml(18) qrc:/FirmwareUpdate.qml:18:5: QML Connections: Cannot assign to non-existent property "onFlashProgress"
[10:14:49:564] ping.devicemanager[Debug]: Start protocol detector service.

patrickelectric, 2019-11-05:
Hi @newJames, sorry for taking so long to reply, I was OOO.

Can you please:
1. Close ping-viewer, check if there is any process running in **Task Manager** and kill it.
2. Open **Registry Editor** and delete ping-viewer settings.
[image]

I just did a fresh install of windows in an empty HD, ping-viewer was able to run without problems here.

If it's still failing after this procedure, please let me know and we can do a teamviewer.

newJames, 2019-11-07:
> Hi @newJames, sorry for taking so long to reply, I was OOO.
>
> Can you please:
>
> 1. Close ping-viewer, check if there is any process running in **Task Manager** and kill it.
> 2. Open **Registry Editor** [...]

Thanks for your answer, I delete the setting and the ping-viewer can work now.   
Your help was greatly appreciated.

closed 2019-11-07 by patrickelectric