bluerobotics/ping-viewer #842: Change P360 ip Address

newJames, 2019-12-02:
Hello！ I want to change the P360 sonar ip Address and port, but I have not found any way to change it,How can I change it ? Thank you

jaxxzer, 2019-12-02:
Hello, the device functions as a dhcp client when it is first powered, and will be assigned an ip address if there is a dhcp server on the network. Changing the static ip address is not presently supported.  Can you let us know why you need to change the ip address?

newJames, 2019-12-03:
I used to connect other sensors by static ip address and controled the ROV designed by mysrlf, and another question is that I tried to connect the p360 sonar by udp stream,but it does not work, I found that your website said Ethernet communications is not supported.

newJames, 2019-12-03:
ok,I just make a dhcp server to connect the only p360 sonar sensor.

patrickelectric, 2019-12-05:
@newJames will this fix your problem ?

newJames, 2019-12-07:
Sure，Thankyou！

closed 2019-12-09 by patrickelectric