bluerobotics/ping-viewer #888: t2.0.6 online displaying pink square

espresso0000, 2020-02-16:
---
## Summary
Version: t2.0.6

Operating System: Windows 10 Pro
What is happening: PingViewer t2.0.6 displaying only pink square in Ping360

## Additional information
[image]

espresso0000, 2020-02-16:
[20200216-165939704.txt](https://github.com/bluerobotics/ping-viewer/files/4211072/20200216-165939704.txt)

patrickelectric, 2020-02-17:
Hi @espresso0000, glad to see you here !
Thank you for your test and feedback, we should have a fix for that in the following days, I'll let you know!

espresso0000, 2020-02-17:
Thank you @patrickelectric , looking forward to testing it out with the new upgrades.  Preparing for a cave dive Mar6-8 with the Ping360 & 1D to see what they can do.

patrickelectric, 2020-02-18:
Hi @espresso0000, We have a new continuous release with the fix: https://github.com/bluerobotics/ping-viewer/releases/download/continuous/pingviewer_release.zip

Can you kindly test it and see if you have the same problem ?
If so, please share the GUI log.
You are using a different video card from us, so the log is really helpful to understand and fix any problem for you : )

espresso0000, 2020-02-18:
@patrickelectric The fix was successful, Ping360 functional, chart rotation also confirmed functional.  Selecting the Ping1D crashed a few times before it functioned.  Attached is 2 gui log files utilizing the continuous release.

[20200218-124427182.txt](https://github.com/bluerobotics/ping-viewer/files/4220858/20200218-124427182.txt)
[20200218-125304677.txt](https://github.com/bluerobotics/ping-viewer/files/4220863/20200218-125304677.txt)

jaxxzer, 2020-02-25:
The patch #889 had the opposite effect for me:
adding the line `precision mediump float` gives a pink square.

jaxxzer, 2020-02-25:
```
[18:50:32.680] default[Warning]: QOpenGLShader::compile(Fragment): 0:8(1): error: syntax error, unexpected NEW_IDENTIFIER

[18:50:32.680] default[Warning]: *** Problematic Fragment shader source code ***
#ifdef GL_KHR_blend_equation_advanced
#extension GL_ARB_fragment_coord_conventions : enable
#extension GL_KHR_blend_equation_advanced : enable
#endif
#define lowp
#define mediump
#define highp
#line 1
/**
 * Based over: https://gist.github.com/KeyMaster-/2bb5e20f824241f3caef
 * Tilman Schmidt [KeyMaster-] example
 */

// Set float precision
precision mediump float;

uniform sampler2D src;
uniform float angle;
varying vec2 coord;

void main() {
    vec2 sizeOverRadius = vec2(2.0, 2.0);
    float sampleOffset = 0.0;
    float polarFactor = 1.0;

    //Move position to the center
    vec2 relPos = coord - vec2(0.5 ,0.5);

    //Adjust for screen ratio
    relPos *= sizeOverRadius;

    //Normalised polar coordinates.
    //y: radius from center
    //x: angle
    vec2 polar;

    polar.y = sqrt(relPos.x * relPos.x + relPos.y * relPos.y);

    //Any radius over 1 would go beyond the source texture size, this simply outputs black for those fragments
    if(polar.y > 1.0){
        gl_FragColor = vec4(0.0, 0.0, 0.0, 0.0);
        return;
    }

    polar.x = atan(relPos.y, relPos.x);

    //Fix glsl origin with src data
    polar.x += 3.1415/2.0;

    //Normalise from angle to 0-1 range
    polar.x /= 3.1415*2.0;
    polar.x = mod(polar.x, 1.0);

    //The xOffset fixes lines disappearing towards the center of the coordinate system
    //This happens because there's only a few pixels trying to display the whole width of the source image
    //so they 'miss' the lines. To fix this, we sample at the transformed position
    //and a bit to the left and right of it to catch anything we might miss.
    //Using 1 / radius gives us minimal offset far out from the circle,
    //and a wide offset for pixels close to the center
    float xOffset = 0.0;
    if(polar.y != 0.0){
        xOffset = 1.0 / polar.y;
    }

    //Adjusts for texture resolution
    xOffset *= sampleOffset;

    //This inverts the radius variable depending on the polarFactor
    polar.y = polar.y * polarFactor + (1.0 - polar.y) * (1.0 - polarFactor);

    //Sample at positions with a slight offset
    vec4 one = texture2D(src, vec2(polar.x - xOffset, polar.y));
    vec4 two = texture2D(src, polar);
    vec4 three = texture2D(src, vec2(polar.x + xOffset, polar.y));
    gl_FragColor = max(max(one, two), three);
}

***
[18:50:32.680] default[Warning]: QQuickCustomMaterialShader: Shader compilation failed:
[18:50:32.680] default[Warning]: "0:8(1): error: syntax error, unexpected NEW_IDENTIFIER\n"
```

PR #889 (2020-02-18): opengl: polarplot: Add missing float precision
Some backends does not provide a default precision for fragment shaders.

Fix #888
PR #901 (2020-02-26): opengl: set precision only for GL_ES
inspired by https://github.com/pxscene/pxCore/pull/690

fix #888
closed 2020-02-26 by jaxxzer