bluerobotics/ping-viewer #900: ping1d detected on ping360 port

jaxxzer, 2020-02-25:
[image]

jaxxzer, 2020-02-26:
and vice-versa: 
[image]

BoKoppel, 2020-05-05:
Logfile and event description if this helps
System: Only one Ping360, only one instance running
Software as april 2020: QGC 4.0.5, Companion 0.0.20, Ardusub 4.0.1

Symptom: Pingviewer automatic connection shows only green Ping360, connection OK
Now going back to "devices" there is also an red Ping1D together with green Ping360
Ping360 now hangs and need hard restart by power off.
This hanging does not alway happen...
Temporary solution: do not enter devices after connection.
[20200505-121023809.txt](https://github.com/bluerobotics/ping-viewer/files/4580264/20200505-121023809.txt)

Bo