bluerobotics/ping-viewer #903: Firmware information is not received reliably

jaxxzer, 2020-02-26:
[image]

jaxxzer, 2020-02-26:
we should ask again if we don't get the answer.

danielgud, 2020-04-30:
@bluerobotics/software-team The firmware options never populate on my work computer. Also had a customer that could not update firmware because of this which lead them to try the stm32flash procedure with a bad firmware file.

patrickelectric, 2020-04-30:
Hi @danielgud can you share one of your GUI logs with that problem ?
I'm unable to replicate it here.

patrickelectric, 2020-05-01:
Log from @danielgud 
[20200501-103708440.txt](https://github.com/bluerobotics/ping-viewer/files/4565794/20200501-103708440.txt)

PR #920 (2020-05-01): appveyor: Update to use 64bits openssl 1.1
Fix #903
closed 2020-05-01 by patrickelectric