bluerobotics/ping-viewer #924: Couldn't use NMEA simulator

3nd0y, 2020-05-13:
Hello,

I don't know if it's bug or request, but i think it's bug. Since i don't have echosounder so i use simulator with NMEA ouput $SDDBT and connected via COM port bridge emulator but ping viewer couldn't display it, 

the bridge port is COM1-COM2, COM1 connected to emulator, COM2 to ping viewer. I tried to monitor COM2 ouput using python serial, it gives output as expected but nor in ping viewer.. 

[image]

patrickelectric, 2020-05-13:
Hi @3nd0y, Ping-Viewer only supports [ping-protocol](https://docs.bluerobotics.com/ping-protocol/) for now, but it's totally possible (and easy) to integrate NMEA input on it.
It's only necessary to create a device class for it, and everything else is should be really straightforward.

patrickelectric, 2022-05-27:
Be free to open a PR.

closed 2022-05-27 by patrickelectric