bluerobotics/ping-viewer #927: ping360 latency issue

jaxxzer, 2020-07-16:
tested on v2.1.0
The latency to next profile request on viewer side is erratic, and sometimes way too large, here you can see it's almost 25ms
[image]

here is the capture:
[ping360.zip](https://github.com/bluerobotics/ping-viewer/files/4934480/ping360.zip)

patrickelectric, 2020-11-12:
Can you show a screnshot with both RX and TX, this 25ms can be the time of the sensor to do the sample.
Without the time between what ping-viewer receives (end of RX package) and what ping-viewer sends (start of TX package) the time in the screenshot does not help much. I would recommend to use the serial-TTL output and not the RS485-TTL.
There is also the RS485 switch time that may affect the translation between serial-TTL and RS485-TTL.

patrickelectric, 2022-05-27:
The auto response messages should have fixed this issue since ping-viewer does not probe for messages anymore.

closed 2022-05-27 by patrickelectric