bluerobotics/ping-viewer #942: Playback Key-Bindings

Eliot-Insight, 2020-12-09:
## Summary
Common playback key-bindings for when replaying data. 

## Details
Specifically after something like:
- `SPACE` for play/pause
- &#8594;/&#8592; arrow keys to skip forwards/backwards 10 seconds
- maybe `SHIFT`+arrows to skip a smaller amount (e.g. 1 second or 1 ping)
- ideally also a speed control option to speed up or slow down playback (e.g. `d`/`a` or &#8593;/&#8595; to increase/decrease playback speed multiplier in 0.1x increments)

Eliot-Insight, 2020-12-10:
Just figured out that if you click the play-pause button then it gains focus, and `SPACE` works while focus is maintained. Would be preferable if the scope can be broadened so `SPACE` can toggle play/pause even with the replay menu hidden (so the data at the bottom of a limited sector angle can be visible).

ES-Alexander, 2024-09-26:
Similar to #687, but that one's focused on live control