bluerobotics/ping-viewer #945: How can we get received acoustic signal?

lixinghe1999, 2020-12-17:
## Summary
_A clear and concise description of what you want to happen._
Hi, BlueRobotics

I am using Ping360 Sonar right now, however, only have sonar image is not enough for our project. I am here if there is possibility to change/write some code and get the raw received signal (periodic signal with noises)?

Thanks a lot, I think this feature can make Ping360 much perfect.
## Additional information
_Add any other context or screenshots/mockups about the feature request here._

Eliot-Insight, 2020-12-17:
You might want to have a look at the [ping-python](https://github.com/bluerobotics/ping-python) library, which gives you direct access to the signal, and low-level control of the sonar settings.

If you're happy with the setting control provided by ping-viewer and just want to analyse the signals afterwards you can follow the [decode sensor binary log](https://github.com/bluerobotics/ping-viewer/blob/master/examples/decode_sensor_binary_log.py) example which uses ping-python to extract recorded data from the binary data logs that ping-viewer stores.

lixinghe1999, 2020-12-18:
Have tried Ping-python, it does control the sonar, but I am afraid I still can't have raw data.
The data in ping-protocol is already processed, they are not received signal, but echo strength in different distance. In other words, those data are part of the sonar image.
Will you team support the function of having this raw data in the future?

Eliot-Insight, 2020-12-18:
I don’t work for bluerobotics so can’t say what might get supported because I don’t know. In saying that, my understanding of the ping360 is that the transducer outputs a sound pulse at a set frequency and then measures the strength of the response over time after that pulse. Accordingly the ‘raw data’ is the ‘echo strength’ at each time step. You can get finer time resolution by specifying a shorter measurement period, at the cost of not being able to measure as far because of the maximum number of samples per ping.

I’m not sure what you mean by the ‘periodic signal with noises’ that you said you’ want. If your question hasn’t been answered please try to more clearly describe your understanding of what the ping360 is doing, and specify the point in the ping process where you think data exists that isn’t currently available through the existing interface.

lixinghe1999, 2020-12-18:
Thank you for your reply! I know it is hard to understand because what I want is more than sonar image.

In my understanding, firstly, as you said, the transducer outputs a sound pulse at a set frequency. Then, the receiver get a sound pulse at the same frequency but with some noise and time delay(echo). After some wise processing (like cross-correlation and filter), we can measure the strength of response over time (This is what we have now).
What I want is the echo ( a Mechanical vibration), currently the data have one step more: find out response strength over time.

In below image, the first row is sound pulse, the second row is echo signal (what I want), the last row is response strength (currently we have).
Compared to first subfigure, the second one has been delayed for 1000. So in the third subfigure, the peak of response strength is close to 1000.
[image]

patrickelectric, 2021-01-27:
Hi @lixinghe1999 and @Eliot-Insight, Ping360 only works with raw data, there is no post-processing on the sensor, if you don't want the sensor to act as a transmitter but only as a receiver, there is no way to do it in code, the only way is to open the sensor and do some hardware changes, but that's not recommended and we can support that.

patrickelectric, 2021-01-27:
@lixinghe1999 also, take in mind that this is ping-viewer development repository, your questions is more related to ping360 development, that can be done in our forum: https://discuss.bluerobotics.com/

closed 2021-04-05 by patrickelectric

leenpu, 2022-11-25:
> Thank you for your reply! I know it is hard to understand because what I want is more than sonar image.
>
> In my understanding, firstly, as you said, the transducer outputs a sound pulse at a set [...]

leenpu, 2022-11-25:
> Thank you for your reply! I know it is hard to understand because what I want is more than sonar image.
>
> In my understanding, firstly, as you said, the transducer outputs a sound pulse at a set [...]

Hello!
May I ask whether the three pictures you uploaded are the measured data of ping360?
I have the same problem as you. Have you solved your problem?

leenpu, 2022-11-25:
> Thank you for your reply! I know it is hard to understand because what I want is more than sonar image.
>
> In my understanding, firstly, as you said, the transducer outputs a sound pulse at a set [...]

@lixinghe1999 
Hello!
May I ask whether the three pictures you uploaded are the measured data of ping360?
I have the same problem as you. Have you solved your problem?

lixinghe1999, 2022-11-26:
> @lixinghe1999 Hello! May I ask whether the three pictures you uploaded are the measured data of ping360? I have the same problem as you. Have you solved your problem?

for the 3 figures, they are fake data generated for illustration.
I believe Ping360 has done the processing in hardware, so it is no way to access the original sound but only the response, hope it helps.

leenpu, 2022-11-30:
> for the 3 figures, they are fake data generated for illustration. I believe Ping360 has done the processing in hardware, so it is no way to access the original sound but only the response, hope it [...]

ok,thank you!

Leonidas-Lee, 2023-04-22:
> You might want to have a look at the [ping-python](https://github.com/bluerobotics/ping-python) library, which gives you direct access to the signal, and low-level control of the sonar settings.
>
> If [...]

Hello!
May I ask how to use ping Python?I am using a ubuntu16.04 of VMware, but there is a problem.
[image]

Leonidas-Lee, 2023-04-22:
> Have tried Ping-python, it does control the sonar, but I am afraid I still can't have raw data. The data in ping-protocol is already processed, they are not received signal, but echo strength in [...]

Hello!
May I ask how to use ping Python? I am using a ubuntu16.04 of VMware, but there is 
[image]
a problem.

lixinghe1999, 2023-04-22:
> Hello! May I ask how to use ping Python? I am using a ubuntu16.04 of VMware, but there is a problem. [image]

Sorry, I use Ping-python on Windows, just pip install and the usage is pretty simple.
My advice is trying write some lines of code based on API rather than directly run the example script, which may not be maintained well.

Leonidas-Lee, 2023-04-22:
> Sorry, I use Ping-python on Windows, just pip install and the usage is pretty simple. My advice is trying write some lines of code based on API rather than directly run the example script, which may [...]
ok, What is the compilation environment on Windows?PyCharm or jupyter notebook, or others?

Leonidas-Lee, 2023-04-22:
> You might want to have a look at the [ping-python](https://github.com/bluerobotics/ping-python) library, which gives you direct access to the signal, and low-level control of the sonar settings.
>
> If [...]

and I have one more question，According to your description on the ping360 official website: "https://discuss.bluerobotics.com/t/how-to-scan-as-fast-as-ping-viewer-in-python/9130", may I ask if the receiving transducer is a single transducer or a linear array？

lixinghe1999, 2023-04-22:
> and I have one more question，According to your description on the ping360 official website: "https://discuss.bluerobotics.com/t/how-to-scan-as-fast-as-ping-viewer-in-python/9130", may I ask if the [...]

Just pycharm

Leonidas-Lee, 2023-04-22:
> Just pycharm

ok, thank you! and I have one more question，According to your description on the ping360 official website: "https://discuss.bluerobotics.com/t/how-to-scan-as-fast-as-ping-viewer-in-python/9130", may I ask if the receiving transducer is a single transducer or a linear array？Do you know?

Leonidas-Lee, 2023-04-22:
> You might want to have a look at the [ping-python](https://github.com/bluerobotics/ping-python) library, which gives you direct access to the signal, and low-level control of the sonar settings.
>
> If [...]

Hello!
May I ask how to use ping Python? Do I need to connect to ping360 sonar? Or just read the decode sensor binary log file? What is the specific operation?

Leonidas-Lee, 2023-04-22:
> Hello!
> May I ask how to use ping-python? Do I need to connect to ping360 sonar? Or just read the decode sensor binary log file? What is the specific operation?

Hello!
May I ask how to use ping-python? Do I need to connect to ping360 sonar? Or just read the decode sensor binary log file? What is the specific operation?

Leonidas-Lee, 2023-04-22:
> Sorry, I use Ping-python on Windows, just pip install and the usage is pretty simple. My advice is trying write some lines of code based on API rather than directly run the example script, which may [...]

Hello!
May I ask how to use ping-python? Do I need to connect to ping360 sonar? Or just read the decode sensor binary log file? What is the specific operation?