bluerobotics/ping-viewer #997: MacOS: enable opening multiple application instances (without terminal)

ES-Alexander, 2021-08-04:
## Summary
Allow easily opening a second instance of Ping Viewer on MacOS to enable simultaneously using multiple sensors (or complete #982).

## Additional information
When clicking icons in MacOS the user is taken to an existing open window if there is one. The only way to get around this (for a user) is to use terminal or a shell script that can run `open --new /Applications/pingviewer.app`. Would be great if there was something like a right-click option for the icon in the taskbar that enables opening a `New Window` like is available for Terminal and Finder and similar (unfortunately couldn't find how to do this with Qt, but I likely don't properly know what to search for/where to look).
[image]

Could also/instead enable this with a `File/New Window (⌘ N)` menubar option
[image]

patrickelectric, 2022-05-27:
This appears more as a mac bug (or feature). From what I could search, this is a normal issue on the OS (not our falt). What you see in terminal is a "New Window" and not a  "New Application". For the terminal is a simple application that manages multiple windows via a service, not opening the same application multiple times.
It's indeed something that we could fix... but the OS does not help with that, and a quick investigation did not come with any solution besides some hacks around Mac OS. If the solution did not come that easily, the tradeoff between investigation + fixing does not worth for now doing this task.

closed 2022-05-27 by ES-Alexander

patrickelectric, 2022-05-27:
May help: https://github.com/bluerobotics/ping-viewer/pull/1046