bluerobotics/ping-viewer #1015: Ping360: hide bootloader

jaxxzer, 2021-09-16:
[image]

Set the ping360 ip to 0.0.0.0, and it will restart into the bootloader, which has it's own (undocumented) ip.

patrickelectric, 2021-09-17:
Ok, so the idea here is to not allow the user to set 0.0.0.0 ip address on ping360 ?

ES-Alexander, 2022-03-28:
@patrickelectric
> not allow the user to set 0.0.0.0 ip address on ping360 ?

That seems problematic. I think the presented idea was to hide the bootloader connection (`192.168.0.111`), since it can be confusing to the user and it allows attempting to configure it, even though it doesn't actually support configuration.

I think this is a bit less essential since the blue "Configuration is required" status now comes up for the one you're actually intended to configure, but it could still be helpful to disable IP configuration for the bootloader, and/or put in the status message that it's suspected to be a bootloader, and not the user-accessible device.