bluerobotics/ping-viewer #1033: Re-Encoded Replay Display Issues

ES-Alexander, 2022-01-17:
## Summary
Version: 2.3.1

Operating System: Mac (Monterey)

What is happening: As per PR #1029, where I implemented a binary log file encoder,

> I've tested the encoding and it seems to work properly (the output can be viewed by PingViewer), although depending on how many/which messages are included, sometimes the output doesn't display.

How to reproduce it: Open PingViewer, record a binary log, decode it with Python, remove/filter out some messages, and re-encode, then try to re-play the re-encoded file with Ping Viewer (works fine with some re-encoded files, but fails to display some others).

## Additional information

> **NOTE: @patrickelectric was not able to reproduce this with the same files**

I think the display inconsistency is most likely a Ping Viewer bug because I was writing only `2300` messages (and only `1300` messages for a Ping1D file), and depending on the start and stop points and step it would sometimes work and sometimes not. As I understand it, if the encoding works for one set of `2300` messages then it should also be fine with other ones - especially the same set of messages just with a few more/less included.

## Ping360
[ping360_F-fail_P-pass.zip](https://github.com/bluerobotics/ping-viewer/files/7767140/ping360_F-fail_P-pass.zip)

| test settings (Ping Viewer 2.3.1, and 2.3.0) | replay display works? |
| --- | --- |
| original file | no |
| processed file (filter to just `2300` (`device_data`) messages) | yes |
| start at 20th `2300` message | yes |
| start at 30th `2300` message | yes |
| start at 20th `2300` message and skip every 2nd one | yes |
| start at 30th `2300` message and skip every 2nd one | no |

## Ping1D
[ping1d_F-fail_P-pass.zip](https://github.com/bluerobotics/ping-viewer/files/7767246/ping1d_F-fail_P-pass.zip)

| test settings | replay display works? |
| --- | --- |
| original file | yes |
| processed file (filter to just `1300` (`profile`) messages) | no |
| start at 2nd `1300` message | yes |
| start at 10th `1300` message | no |
| start at 20th `1300` message | no |
| start at 30th `1300` message | yes |

patrickelectric, 2022-02-07:
@ES-Alexander Can you check the latest master branch ?

ES-Alexander, 2022-02-08:
> Can you check the latest master branch ?

Still had some issues with both Ping1D and Ping360 files. Used the macos artifact from [here](https://github.com/bluerobotics/ping-viewer/actions/runs/1808243710), which I'm assuming is the correct one to be testing with.

Here's a video and the corresponding GUI log file:

[20220208-214215830.txt](https://github.com/bluerobotics/ping-viewer/files/8023022/20220208-214215830.txt)

https://user-images.githubusercontent.com/25898329/152972095-9549359f-373b-4302-8581-35cbae3f5195.mov

PR #1038 (2022-02-14): Fix parser problem using latest version of ping-cpp 
Fix #1033
closed 2022-02-14 by patrickelectric