bluerobotics/ping-viewer #1034: Make Connecting Smarter

ES-Alexander, 2022-01-17:
## Summary
- Auto-connection could by default connect to the last successful connection, instead of starting in search mode by default
- Manual connection could provide better default ports when selecting Ping1D (9090) or Ping360 (9092) device types. Custom/dev IP and port should be available, but shouldn't come before defaults for standard users.

## Additional information
If user changed the IP subnet last time they used Ping Viewer, they quite likely want to keep that subnet next time. (I'm a bit confused here, I thought Ping Viewer searched for devices on the current computer's subnet rather than a hardcoded one, but maybe not - may need confirmation)

Current manual connection default configuration is set to `192.168.2.2:1234`, which is the default Companion IP (reasonable expectation) and the dev port (less desirable). Current implementation uses the same Entry widget, and hence same value regardless of device type, but ideally specifying a different device type would also change the default connection info, since different devices have different default ports.

> [Relevant forum comment](https://discuss.bluerobotics.com/t/ping-1d-can-not-connect/11330/5):
> You could do a dropdown list on the port section: Ping1D (9090), Ping360 (9092), Custom.
I don’t know how many of us change the IP address of the ROV but even if the PingView does not check the local address mask (in my example 192.168.4.x), it would be great if it instead of always starting in search mode, connects to the last successful connection.
> -Grega (@tunadiver)

patrickelectric, 2022-01-18:
@ES-Alexander I'm not sure if I follow correctly, ping-viewer saves all previous manual connections to run on the search loop.
Or: If a user add a manual connection and close ping-viewer, the next time ping-viewer will use the previous manual connection configuration on the search loop and it should appears as a valid result.

ES-Alexander, 2022-03-28:
@patrickelectric 
> ping-viewer saves all previous manual connections to run on the search loop.

Are you saying that already happens, or asking if that's the desired behaviour? If it doesn't already happen, that's the desired behaviour. The second option doesn't make much sense (to me at least) - you can't pretend something is a valid result if it's actually not, right?

patrickelectric, 2022-05-26:
https://github.com/bluerobotics/ping-viewer/blob/master/src/devicemanager/devicemanager.cpp#L121
https://github.com/bluerobotics/ping-viewer/blob/master/src/sensor/protocoldetector.cpp#L49