bluerobotics/ping-viewer #1053: Source code directly compiled error

leenpu, 2022-11-30:
[image]
Hello! The source code I downloaded on github will make an error if I compile it directly with QT？

patrickelectric, 2022-11-30:
Follow our build instructions here:
https://docs.bluerobotics.com/ping-viewer/build-instructions/

For windows, you need to build it in release mode or remove the kcrash dependency.

patrickelectric, 2022-12-01:
Ok, did you found any problem with msvc2019 ?

leenpu, 2022-12-02:
> 

yes, qt5.15.2 has an error while compiling, It shows that no cmake_cxx_compiler could be .......

patrickelectric, 2022-12-02:
You should be able to use msvc2019.

Please provide the full error log of your compiler output.

leenpu, 2022-12-02:
> You should be able to use msvc2019.
> 
> Please provide the full error log of your compiler output.

Unfortunately, an error occurred when compiling this source code！ the full error log of your compiler output:
[image]
[image]

patrickelectric, 2022-12-02:
You are compiling in debug mode, where KCrash is necessary, as said previously, please build in release mode for windows.

leenpu, 2022-12-03:
> please build in release mode for windows

Thank you, build in release mode for windows， it compiled through except for three warnings！
[image]
[image]

patrickelectric, 2022-12-04:
I don't understand the text on the last two warning. The first one is new.

closed 2022-12-04 by patrickelectric