bluerobotics/ping-viewer #1065: Firmware update failing on Windows 10 (stm32flash 0.5, Failed to init device.)

dkim1113, 2023-10-16:
# **REMOVE THIS SECTION AFTER FINISHING THE BUG REPORT**
- **Bug reporting involves responsibility**
- **Before you report a bug, please make sure that you are:**
    - Willing to provide information to replicate the issue.
    - Willing to correspond with the developers about the issue, potentially over an extended period of time.
    - Prepared to report additional information.
    - If those points all apply to you, then we can move on! Let's file a bug !
- **Be sure that is a real bug**
    -  Real bugs involve crashes, data loss, features that don't work, user interface anomalies, and inconsistencies in behavior.
- **Make sure it hasn't already been fixed**
    - Using relatively recent versions of ping-viewer. Check for new versions [here](https://github.com/bluerobotics/ping-viewer/releases).
- **Make sure you can reproduce it**
    - **Please do not report bugs that you cannot reproduce.**

---
## Summary
Version: 2.4.1

Operating System: Windows 10

What is happening: 
Firmware update failing on Windows 10 (stm32flash 0.5, Failed to init device.)
The error message:
stm32flash 0.5
http://stm32flash.sourceforge.net/
Using Parser : Intel HEX
Interface serial_w32: 57600 8E1
Failed to init device

How to reproduce it: 
Connect Ping2 Sonar to Windows 10 laptop USB port > Open PingViewer > Connect either   baud 9600 or 115200 > click the gear icon, try automatic update (current firmware version: 3.28, latest version: Ping-V3.29_auto.hex OR try manual update after downloading Ping-V3.29_auto.hex file. > Click FIRMWARE UPDATE.

## Additional information
> I am using BLUART USB to TTL Serial and RS485 Adapter (https://bluerobotics.com/store/comm-control-power/tether-interface/bluart-r1-rp/)
> 
[image]

> If possible, add the cra_sh report or the GUI log (Documents/PingViewer/GUI_Log/date-time.txt)
[20231016-144000098.txt](https://github.com/bluerobotics/ping-viewer/files/12912654/20231016-144000098.txt)

ES-Alexander, 2024-03-11:
Someone also ran into this in 2.4.2, as brought up in [this forum post](https://discuss.bluerobotics.com/t/ping1d-firmware-manual-update-fail-via-ping-viewer-both-on-windows-ubuntu/16224).

@patrickelectric any ideas on this?

joaoantoniocardoso, 2024-03-11:
Maybe it needs to be run as admin?

patrickelectric, 2024-03-11:
@joaoantoniocardoso by default ping-viewer run as admin in windows.