bluerobotics/ping-viewer #1074: Ability to record data from MAVLINK stream along with sonar data

sp-ex, 2024-03-13:
It would be very convenient to have an option to record selected data from the vehicle's MAVLINK UDP stream (e.g. from BlueBoat) like GPS position, GPS time, roll, pitch, heading etc. together with each sonar profile. This would help to analyze the profile data in relation to position, i.e making maps of bottom or sub-bottom features, correcting for angular deviation caused by waves or determining the sonar's boresight angles.

ES-Alexander, 2024-05-22:
This is similar to #1009, which has some relevant discussion points.

I can see why recording GPS data would be useful, but that also seems somewhat out of scope for Ping Viewer unless/until it's capable of actually doing something with that data.

Our Ping Viewer development efforts are currently focused on rewriting it with a more modern technology stack, partially to allow it to more effectively integrate with BlueOS and Cockpit, and I expect one of the side-effects of that will likely be more straightforward alignment and replays of the different data streams :-)

In the meantime you might be interested in the [pingSurvey](https://github.com/vshie/pingSurvey) BlueOS Extension, and you could consider creating a modified version that records the information you're interested in. Since that runs onboard the vehicle, it also doesn't require an active connection to Ping Viewer, which is likely useful for BlueBoat missions, especially if they're running autonomously.

ES-Alexander, 2024-09-26:
> I can see why recording GPS data would be useful, but that also seems somewhat out of scope for Ping Viewer unless/until it's capable of actually doing something with that data.

I think position-aligned data on maps for sidescan and bathymetry purposes and the like may be out of scope for Ping Viewer, but there are still some useful and relatively simple use-cases for GNSS data described in #1087, which would presumably be useful during replays as well.