bluerobotics/ping-viewer #1076: Pingviewer

KoffiNaznof, 2024-04-20:
We built an Unmanned Surface Vehicle using Pixhawk and Ping Sonar. The ping sonar will send the signal and generate my image /data through ping_viewer. However, my  Pingviewer shows only a blank screen with "0.00m    Confidence :0%.
Please, what am I doing wrong? What am I supposed to do?
All suggestions are welcomed and appreciated.

patrickelectric, 2024-04-20:
Hi @KoffiNaznof, how are you doing the bridge between ping and ping-viewer ?

KoffiNaznof, 2024-04-20:
I am using BLUART USB to TTL Serial and RS485 Adapter.

KoffiNaznof, 2024-04-20:
I appreciate your quick reply.

patrickelectric, 2024-04-20:
Are you using BlueOS and connect bluart on it or directly in your computer
that ping viewer is running ?

KoffiNaznof, 2024-04-21:
Please, what is BlueOS? and how can access it?

KoffiNaznof, 2024-04-21:
[image]

KoffiNaznof, 2024-04-21:
> Are you using BlueOS and connect bluart on it or directly in your computer that ping viewer is running ?
> […](#)

I connected the bluART directly to the computer . Is there any alternative way of doing it ? I am not too sure about BlueOS.
Thanks

ivnat0, 2026-02-20:
Ya facing the same problem, how to resolve it!?

patrickelectric, 2026-02-20:
> Ya facing the same problem, how to resolve it!?

Can you explain what is your problems ?