bluerobotics/ping-viewer #1077: Vehicle overlay shapes

rov-rob, 2024-04-24:
## Summary
Hi.
Id like to have the option to add custom vehicle shapes as overlay in  pingviewer as this would be of great help operating large vehicles.

- Load in a list of coordinates outlining the vehicle shape. 
    - Fileformat could be something simple like a .csv with x and y
    - Option so set colour and have multiple parts on the vehicle outline.
- Have the overlay shape scale with the selected sonar range.
- Have the overlay crop with the sonar sector setting.

I do not have enough programming knowledge to even know where to look to implement this my self.

## Additional information
Example .csv and mockup pictures so vehicle is not to scale. Assuming ping360 as CRP in the example.

```
#x,y,name,colour
-0.4,0.5,base,blue
0.4,0.5,base,blue
0.4,-1.5,base,blue
-0.4,-1.5,base,blue
-0.4,0.5,base,blue
		
0.3,-1.5,backpack,red
0.3,-1.8,backpack,red
-0.3,-1.8,backpack,red
-0.3,-1.5,backpack,red
```

[image]

[image]

joaoantoniocardoso, 2024-04-24:
Interesting idea. To add on that, I believe a .png (vectorized) would be a more universal approach, maybe with an offset coordinate pair to help centralize.

patrickelectric, 2024-04-24:
Yep, a SVG would be the way to go for sure!

rov-rob, 2024-04-24:
From my experience with survey softwares like qps qinsy, you would define your vehicle shape with a simple csv or cad file(dxf/dwg of similair).

Line width and line colour could then be set by the user in the software without having to change and load different files.

Reading up a bit on svg file format, maybe the stroke and stroke-width can be changed by changed by the user in pingviewer.

ES-Alexander, 2024-09-26:
Vaguely related to #1087